arduino-libraries/WiFi101 #229: WiFiUdpSendReceiveString fails partially if modified for bi-directional transmission.

RamonRibes, 2018-05-10:
Hi, i've modified the sketch to allow communication between 2 MKR1000 boards. Instead of receive external data and sending just an acknowledge back, i wish active communication between them.
So, for testing purposes, i modified the original sketch the less possible to achieve this purpose.
The modified testing sketch is as follows:
```
#include <SPI.h>
#include <WiFi101.h>
#include <WiFiUdp.h>

int status = WL_IDLE_STATUS;
#include "arduino_secrets.h"
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;        // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                 // your network key Index number (needed only for WEP)

const IPAddress arduinoLocal(192, 168, 1, 171);    // MKR1000's local IP.
const IPAddress arduinoRemote(192, 168, 1, 172);   // MKR1000's remote IP.
const IPAddress dns_gateway(192, 168, 1, 1);       // Net DNS & gateway.
const IPAddress subnet(255, 255, 255, 0);          // Subnet mask.
unsigned int localPort = 2390;      // local and remote ports to listen on

const unsigned long maxTimeToSendUDP = 1000L;   // Time to send a UDP Packet.
unsigned long timeLastUdpWasSend = 0L;          // Time the last UDP Packet was send.
int deliveryNumber = 0;                         // Packet number.

char packetBuffer[255];                     //buffer to hold incoming packet
char  ReplyBuffer[] = "acknowledged";       // a string to send back

WiFiUDP Udp;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to WiFi network:
  while ( status != WL_CONNECTED) {
    WiFi.config(arduinoLocal, dns_gateway, dns_gateway, subnet);
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }
  Serial.println("Connected to wifi");
  printWiFiStatus();

  Serial.println("\nStarting connection to server...");
  // if you get a connection, report back via serial:
  Udp.begin(localPort);
}

void loop() {
  unsigned long actualTime;

  // if there's data available, read a packet
  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    Serial.print("Received packet of size ");
    Serial.println(packetSize);
    Serial.print("From ");
    IPAddress remoteIp = Udp.remoteIP();
    Serial.print(remoteIp);
    Serial.print(", port ");
    Serial.println(Udp.remotePort());

    // read the packet into packetBufffer
    int len = Udp.read(packetBuffer, 255);
    if (len > 0) packetBuffer[len] = 0;
    Serial.println("Contents:");
    Serial.println(packetBuffer);

    if (String(packetBuffer) != String(ReplyBuffer)) {   // If received packet is not an acknowledge...
      // send a reply, to the IP address and port that sent us the packet we received
      Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
      Udp.write(ReplyBuffer);
      Udp.endPacket();
    }
  }
  actualTime = millis();
  if (actualTime - timeLastUdpWasSend >= maxTimeToSendUDP) {
    sendUdpPacket();
    timeLastUdpWasSend = actualTime;
  }
}

void printWiFiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

void sendUdpPacket() {
  char bufferOfChars[255];

  //Sending our own Packet:
  deliveryNumber++;
  Serial.print("Sending Packet n# ");
  Serial.println(deliveryNumber);
  sprintf(bufferOfChars, "Packet n# %d", deliveryNumber);
  Udp.beginPacket(arduinoRemote, localPort);
  Udp.write(bufferOfChars);
  Udp.endPacket();
}

```
The starting sequence is as follows:
1. Start the first MKR1000 board with this sketch loaded.
2. Let it connect to the router and see how it is sending udp packets.
3. Start the second MKR1000 board with this sketch modified (just changed the IP addresses from/to local/remote):
```
const IPAddress arduinoLocal(192, 168, 1, 172);    // MKR1000's local IP.
const IPAddress arduinoRemote(192, 168, 1, 171);   // MKR1000's remote IP.

```
4. The first board starts receiving packets, but the second one just sends its own.
5. If i press the reset button on the second board, nothing changes (also tested disconnecting & re-connecting).
6. If i press the reset button on the first board, then both boards start sending/receiving simultaneously.
7. This behaviour repeats always.

I've also tested the AP mode, in this way:
1. Start the first MKR1000 board with this sketch loaded (just changed  `status = WiFi.begin(ssid, pass);` for this:  `status = WiFi.beginAP(ssid, pass);` on this board). Let's name it the 'WifiServer' board.
2. Let it create the wifi network and see how it is sending udp packets.
3. Start the second MKR1000 board with this sketch modified (just changed the IP addresses from/to local/remote):
```
const IPAddress arduinoLocal(192, 168, 1, 172);    // MKR1000's local IP.
const IPAddress arduinoRemote(192, 168, 1, 171);   // MKR1000's remote IP.

```
Let's name it the 'WifiClient' board.
4. The first board starts receiving packets, but the second one just sends its own.
5. The behaviour changes when i start first the WifiClient board: it keeps trying to connect to the non-created-yet network, until i start the WifiServer board. In this moment, both boards start sending/receiving simultaneously. No need to restart the WifiServer one.

Well, i tryed to describe in detail the issue, in order to ease others to test it.
It would be interesting if this could be tested on others boards / systems.
I'm at your disposal for further explanations & tests, if needed.
Thanks!

sandeepmistry, 2018-07-23:
@Rocketct could you please add this to your list of things to look at this week.

Rocketct, 2018-08-07:
Hi @RamonRibes, i had reproduce your issue, after some test i note that it happen only when the second board is turned on after that the first  had start to transmits, what is happen that the second board is able to receive but are not able to send the packet to the first, i had sniffed the packet with wireshark and the only ones packet that transit  in the network are the one of the first board that is turned off.

RamonRibes, 2018-08-07:
Rocketct,
Thanks for your tests.
By reading your response, it seems we agree in the behaviour of the boards, and the issue happens again and again.
So, i think this issue is undesirable, and it's time to someone in charge to take this problem into account, in order to fix it in future upgrades.
Of course, if you need my help in testing something else, i'm at your disposal.
Thanks again!

Rocketct, 2018-08-21:
@RamonRibes thanks for your availability, let me inform you that we suppose that the problem could be in the firmware of the module, i have opened a support request to microchip for a firmware update.

sandeepmistry, 2018-08-21:
@Rocketct please add the case number here.

Rocketct, 2018-08-22:
@sandeepmistry Yes the case number is 00323698 .

Rocketct, 2018-09-27:
Hi @RamonRibes we are following to work on this issue, was released a new firmware version 19.6.1 but this still not solve the issue, we are follow to work for solve it.

RamonRibes, 2018-09-27:
Ok, @Rocketct, thank you!

sandeepmistry, 2018-11-14:
Hi @RamonRibes,

We've received a workaround suggestion from Microchip, could you please try the changes in pull request https://github.com/arduino-libraries/WiFi101/pull/257 and provide your feedback?

RamonRibes, 2018-11-14:
I'll try as soon as possible...