arduino-libraries/WiFi101 #230: WifiClient is littering memory

KugelschreiberY, 2018-05-18:
Hi there,
currently I try to connect to a local web server running on my machine using WIFI101's WebClient. I start a new request in every iteration, unfortunately my micro controller is running out of memory after only a few iterations i.e. it is decreasing in the following manner:

```
18979
11227
3475
```
The code I call every iteration is the following:

```
WiFiClient client;

if (client.connect(server, 4000)) {
    Serial.println("connected to server");
    // Make a HTTP request:

    client.println("GET /value");

    client.println();
}
```
where client it should get the source code of the page `http://server:4000/value`, which is a json string. I know, it is probably not a nice style to get my json string like above, but it is only for a small prototype.

The line of interest is `client.println("GET /value");` when do not include this line, the code is running smoothly. I already added the `client.flush();` and  `client.stop();` at the end of every loop, but it didn't work either.

Anyone has an idea what else could work? Thanks in advance.

klausj1, 2018-05-19:
Hi,

I am not sure what your sketch is doing, and what you mean with "when I do not include this line, the code is running smoothly". Without this line, you do not create an HTTP-Request, so no json-String will be returned. Please provide more information about your sketch, and about what is working and what not.

Did you already have a look at some Arduino HTTP-Library, e.g. https://github.com/arduino-libraries/ArduinoHttpClient? That would make things much easier.

Regards,
Klaus

KugelschreiberY, 2018-05-21:
Thanks for the reply. I was testing out also other HTTP libraries, the error occurred in all of them. Then I found out that the error occurred in a self-written "append" function for character arrays. After fixing that bug, everything runs as it should. 

My bad, sorry for the inconvenience!

closed 2018-05-21 by KugelschreiberY