arduino-libraries/WiFi101 #233: mkr1000 ap udp multiple connections

PieroProcessing, 2018-06-03:
HI, 
I started with mkr1000 an access point to send udp data. it works fine but after one connection is established I can't connect any other machine through wifi on it. Is it a common problem?

sandeepmistry, 2018-07-23:
Hi @PieroProcessing,

This is related to https://github.com/arduino-libraries/WiFi101/issues/46.

You'll need to call `.begin(...)` on the `WiFiUDP` instance again, as the firmware doesn't automatically bind the sockets when a connection is lost.

sandeepmistry, 2019-09-18:
Hi @PieroProcessing, did you have a chance to try my suggestion?

PieroProcessing, 2019-09-19:
I found a solution. It s pretty raw but it works.
https://github.com/PieroProcessing/MKR1000-Osc-Upd/blob/master/WiFiUdpSendReceiveRawOscMkr1000.ino

sandeepmistry, 2019-09-19:
Ok, then let's close this for now.

closed 2019-09-19 by sandeepmistry