arduino-libraries/WiFi101 #236: Add simple way to reset provisioning

Shpagooder, 2018-06-19:
At the moment, I see no way to reset/delete the data that has been stored using wifi101 provisioning so that the provisioning AP can re-open and new credentials can be inputted.

sandeepmistry, 2018-07-23:
Hi @Shpagooder,

You can reflash the firmware to wipe the stored provisioning values.

https://www.arduino.cc/en/Tutorial/FirmwareUpdater

Shpagooder, 2018-07-23:
Thanks for the reply! I'm aware you can do that, but that's a very impractical solution. I assume it would be a rather easy feature to implement since you are already reading/writing to flash when using provisioning. Some applications need to have access to the provisioning data in order to change the SSID or password without having to re-flash the firmware.

sandeepmistry, 2018-07-23:
Unfortunately, I don't think it's trivial to implement, we'd have to reboot WINC1500 in download mode, erase the relevant flash page and reset the WINC1500 to normal mode. As from looking at [m2m_wifi.h](https://github.com/arduino-libraries/WiFi101/blob/master/src/driver/include/m2m_wifi.h) there are no API's to clear provisioning data.

We would welcome pull requests for this feature however.

sandeepmistry, 2019-09-18:
We don't have any plans to implement this feature at this time, so I'm closing and marking as won't fix for now.

closed 2019-09-18 by sandeepmistry