arduino-libraries/WiFi101 #241: DHCP options

blainemulugeta, 2018-09-19:
Is there anyway to configure the DHCP options sent in the DHCP transaction? 

Example: is there a DHCP client function that set's those options? or a file ... please help trying to add DHCP option 161 to DHCP DISCOVER message on the Adafruit Feather M0..any insight would be greatly appreciated! 

Thank you!

sandeepmistry, 2018-09-20:
Hi @blainemulugeta,

I don't believe the firmware on the WINC1500 supports this feature. I suggest you contact Microchip technical support by creating a new support ticket on the following page: http://www.microchip.com/technical-support/hot-topics

sandeepmistry, 2019-09-18:
We have no plans to support this feature at this time, so I'm closing and marking as won't fix for now.

closed 2019-09-18 by sandeepmistry