arduino-libraries/WiFi101 #251: Wifi client hangs after many unsuccessfull connection attempts.

geekfactory, 2018-10-18:
Hello.

While developing an application for a client (arduino MKR1000), I discovered an issue which I think it's related to the wifi library or the firmware on the ATWINC1500. The behavior of the system is as follows:

-I have code that performs periodic HTTP requests over SSL (WiFiSSLClient).
-We ran a test that consisted in the simulation of failure of internet connection. For this test we ran the wireless access point with the Ethernet cable that connects it to the internet unplugged. Prior to the disconnection of the Ethernet cable, the code works as expected and communication with server is ok.
-The code repeatedly tries to connect to the server using client.connect(), even when we don't expect it to succeed because the Ethernet cable disconnected makes the server unreachable.
-When we reconnect the ethernet cable of the access point, the first call to client.connect() actually returns SUCCESS so our code starts using the socket.
-Then the code tries to send data (http headers) to the server using client.print() and client.println() but it never returns from the call. 

Has anybody else reported this behavior?

Thank you.

Rocketct, 2018-10-19:
Hi @geekfactory could you please share a minimal sketch that reproduce the issue?

geekfactory, 2018-10-21:
Hi @Rocketct, thank you for your reply.

I´m trying to reproduce the issue with a simpler sketch, but I have been unable to reproduce it. Instead I´m now suspecting of my NTP library because on its own it causes the arduino to crash when the network returns from an outage, this is the only crash I´m able to reproduce on a simple sketch.

I disabled the NTP update code and the program now seems to run reliably. But i´m suspecting on some trouble on the  DNS client that resolves hostnames to IP addresses as it´s used by both the HTTP client and the NTP code.

For the record: I´m using the latest available version of the WiFi 101 library 0.15.2 and latest firmware for the WINC1500 (19.5.2).

I´ll post the simplified code and the NTP code as soon as I have it ready.

Rocketct, 2018-10-29:
well @geekfactory, ok i wait for your updates

S2Doc, 2019-01-23:
I have had very similar problem sending MQTT packets through an access point that has lost its connection to the internet at large.  This generally fails gracefully but hangs indefinitely every few thousand attempts (no fixed number).  When it hangs, it is in the WiFiClient::connect().
I should note that I also call the NTP server between sending packets.

Using Adafruit M0 WINC1500

sandeepmistry, 2019-01-30:
Closing this for now due to lack of feedback, please re-open with requested info if you are still interested.

closed 2019-01-30 by sandeepmistry