arduino-libraries/WiFi101 #256: SSL Verification Bypass help needed

gjt211, 2018-11-06:
After some digging (and yes I have no idea what I am doing) I have discovered a flag that should allow me to use SSL but bypass the certificate verification. The flag name is `SO_SSL_BYPASS_X509_VERIF` and I found it in socket.c line 1167.

What I would like to know is how to set this flag in my Arduino code.
I am also wondering if this will actually work?
By this I mean will the SSL connection will be successful without having the certificate in the WINC1500?
Yes I am aware of the security issues this raises but at this point in time that is not relevant to what I need.

Any help is appreciated.

gjt211, 2018-11-13:
Any ideas anyone please?

sandeepmistry, 2019-09-18:
Hi @gjt211,

I suggestion you contact Microchip/Atmel support for this question.

closed 2019-09-18 by sandeepmistry