arduino-libraries/WiFi101 #258: Ongoing WINC hangs with SPI errors

desvdp, 2018-11-20:
HI All, 

for a while I'm fighting an issue where the WiFi locks up after 1 to 5 hrs, the board I'm using is an Adafruit M0 WiFi. The behavior is easy to describe: after said time the orange (network) led remains continously on, the RSSI reports 0, the network becomes unreachable but the normal (user) loop continues without issues. Insterting a routine (found on adafruit)  to check free memory, is doens't look like a heap-stack collision. 

When I enabled the WiFi101 debugging I got: 

```
APP)(INFO)(APP)(DBG)[�O␁][83462](APP)(INFO)recv callback session ID = 536903164
(APP)(INFO)(APP)(DBG)[�O␁][83462](APP)(INFO)recv callback session ID = 536903284
... [ repeated many times ]... 
(APP)(INFO)(APP)(DBG)[�O␁][83462](APP)(INFO)recv callback session ID = 536903164
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data response read...(20007d34)
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data read...
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)Reset and retry 536903012 149fe
... [repeated about 10 times ] 
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data response read...(20007d34)
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data read...
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)Reset and retry 536903012 149fe
(APP)(INFO)(APP)(ERR)[�E␁][82481](APP)(INFO)Bus error (5).536903076 145d6
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data response read...(20007d34)
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data read...
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)Reset and retry 536903012 149fe
... [ repeated about 20 times] ... 
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data response read...(20007d7c)
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)[nmi spi]: Failed data read...
(APP)(INFO)(APP)(ERR)[�G␁][82481](APP)(INFO)Reset and retry 536903084 149fe
(APP)(INFO)(APP)(ERR)[�O␁][82481](APP)(INFO)(HIF)Fail to wakup the chip
```

No other devices are conneced to the the SPI bus the WiFi chip is on, so I have absolutely no explanation for this odd "bus" behavior, the signal integrity also looks fine. 

The total sketch does use quite some other M0 peripherals like: 
- a timer on TC5 with an pin write in the ISR
- a PWM signal based on TCC0
- an incoming interrupt on PA5 (A4) 
- a second SPI bus using sercom1 (pins PA16,PA18, PA19) connected to an FRAM chip
- the default I2C bus on sercom3 connected to a port expander

Software wise: 
 - contrary to the default example, the network functionality lives in a seperate class wich is instantiated once on the stack rather then being fully defined in the main loop
- also I'm using the WiFi101OTA lib (removing this doens't seem to impact; however the WiFi trace above is highly likely  the result of mDNS packets coming in) 

When the default example is loaded the (like the simple web server) the system does seem to hold for a longer time (too long to put a number on it). I've been trying to swap in/out pieces of code, but so far without success. 
Has anyone seen a similar behavior with regards to the bus errors; or does this sound familiar to you. 

Help is greatly appreciated, as I'm running out of ideas ... 
Thanks for spending some brain cycles on this one !

sandeepmistry, 2019-09-18:
Hi @desvdp, I suggest you contact Microchip/Atmel support for further assistance with this.

Please add a comment with your ticket number if you decide to contact them.

sandeepmistry, 2019-09-18:
Also, since you are using Adafruit h/w please contact their support or forum. Hopefully someone from their team can investigate further and provide a pull request to resolved your issues.