arduino-libraries/WiFi101 #261: arduino 1.8.8 firmware updater does not show new firmware version

novski, 2018-12-10:
My arduino firmware updater from Arduino1.8.8 does not show the newest firmware version.
Im running a firmware uploader skript from the examples for adafruit Feather M0 and walked through [this tutorial](https://learn.adafruit.com/adafruit-feather-m0-wifi-atwinc1500/updating-firmware). 
I have updated the Library to Wifi101 0.15.3.
But the dropdown for firmware version is only populated with one version and that seams outdated. Its called: WINC1501 Model A (19.**4**.4) (Arduino Wifi 101 Shield). The firmware checker does not pass because the newest firmware version checked for is 19.**5**.4.  it recognizes the 19.**4**.4.
How can i get the newest firmware version?

rsispal, 2018-12-15:
I found the same thing with a number of WINC1500 breakouts. For this firmware upgrade process, I downgraded Arduino IDE to 1.8.7 and then use 1.8.8 for actual development.

sandeepmistry, 2019-09-18:
This should be solved with the latest firmware updater version.

closed 2019-09-18 by sandeepmistry