arduino-libraries/WiFi101 #268: getTime always returns 0

marco-mo, 2019-02-13:
Hello,
I'm experimenting with an Arduino MKR1000 and the WiFi101 library. I'm facing a problem using the getTime() function since it returns always 0. 
Here are the checks that I already made:
- I've checked that Arduino can reach internet, resolve addresses and successfully ping an host on internet both using the name and the address. 
- To be sure I tested the DNS capability with the following instruction (the host should be the one used by the WiFi101library): WiFi.hostByName("time-c.nist.gov", tIP), and got the right address.
- I've checked also that from inside my home network the ntp servers on internet are reachable (I've tried with a NTP query tool on the PC to query the server at time-c.nist.gov).
- I tried also to increase the timeot in the getTime() function in WiFi101.cpp file from 5 to 60 s
- I tried to call explicitely the function m2m_wifi_enable_sntp(1)
Everything gave no results.

My software configuration is the following:
- IDE 1.8.8 
- WiFi101ver. 0.15.3 with latest firmware on Arduino
- Arduino is assigned static IP, gateway and DNS addresses

Could this be a problem in the library?

broddo, 2019-03-04:
This may be a silly answer so my apologies if you've already checked this, but are you waiting long enough? It can take a long time for the built-in NTP functions to get a time value. The reason I ask this is because I also noticed that getTime would return 0 when debugging via JTAG. But if I just let the program run, eventually it would start returning a valid time.

marco-mo, 2019-03-04:
Thank you for your reply.
To get the NTP time I use the following code:

`int InitRtcWithNTP(void)
  {
  unsigned long ulEpoch;
  int iNumberOfTries = 0, iMaxTries = 6;
  
  do 
    {
    ulEpoch = WiFi.getTime();
    iNumberOfTries++;
    }
  while ((ulEpoch == 0) && (iNumberOfTries < iMaxTries));

  if (iNumberOfTries >= iMaxTries)
    {
    PRINTLN("NTP unreachable!!");
    return 1;
    }
  PRINT("Epoch received: ");
  PRINTLN(ulEpoch);
  MyRtc.setEpoch(ulEpoch);
  return 0;
  } `

I willtry using more attempts and send you the results.

marco-mo, 2019-03-11:
Hello!
I've found which is the problem with the library. If I use a dhcp to get the address of the wifi interface, everything works well. If I use a static address, the getTime() returns always 0.
In this code fragment:
    WiFi.config(tLocalIP, tDnsIP, tGatewayIP, tSubnetMsk);
    status = WiFi.begin(ssid, pass);
if I comment out the first row, I can retrieve the time. Leaving the row in place triggers the problem.
I need to use a static ip address. How can I bypass the problem?

broddo, 2019-03-18:
Ok, that definitely looks like a problem. If you are just looking to bypass the problem, could you have the router handle the static IP addresses instead? This would be my preferred method (and network admins will be a lot happier!). It has a lot benefits - if you have multiple devices with static IPs, you now have a central point for managing them all. You can add/delete/reassign IPs etc., and you are not likely to accidentally have two devices trying to claim the same static IP (which could easily happen if you hardcode them). It also saves you from having to hardcode them which would involve either making/compiling different versions of the firmware for each device or including a config file somehow. 

That said - this doesn't actually fix the issue that setting a static IP seems to stop the NTP from getting a time.

marco-mo, 2019-03-19:
Thank you for your advice. I'll use the strategy that you suggest to bypass the problem. 
Do you think that the problem will be solved in the future?

sandeepmistry, 2019-09-18:
Hi @marco-mo, I suggest to contact Microchip/Atmel support for further assistance.