arduino-libraries/WiFi101 #269: ConnectWPA fails on Orbi RBR50 and WNDR4500v3

sjemmett, 2019-03-19:
I have the Adafruit WINC1500 shield and a Arduino Mega 2560R3.  Running the "ScanNetworks" sketch returns the expected networks.  

Running the "ConnectNoEncryption" works fine.  

Running the "ConnectWPA" fails in that it never gets past the "Attempting to connect to WPA SSID: Betelgeuse" phase.  

I have tried this both on an Orbi RBR50 and a WNDR4300v3 with the same results - scanning shows the expected networks, connecting with no password works fine, but trying to connect with WPA fails.

sjemmett, 2019-03-19:
Shud have mentioned that the firmware versions reported by "CheckWifi101FirmwareVersion" is 19.4.4.  This is on the shield that was purchased within the last two weeks.

ogiewon, 2019-03-19:
Which Arduino IDE version are you running?  Try using the WiFi101 firmware updater in version 1.8.6 or 1.8.7, instead of 1.8.8 (or newer, if exists). I noticed that the adafruit WINC1500 doesn’t seem to be getting the firmware updates it deserves in later Arduino IDE versions.

sjemmett, 2019-03-20:
Currently using 1.8.8.  I’ll look for 1.8.6/7, give the updated a try and then try “connectwpa” and provide a response.

sjemmett, 2019-03-20:
Well now, there is joy on my end!  Using 1.8.6, i was able to update the firmware to 19.5.4 and the "ConnectWPA" works correctly.

Thanks

closed 2019-03-20 by sandeepmistry