arduino-libraries/WiFi101 #270: [WiFiSSLClient] can't connect to maker.ifttt.com:443

sandeepmistry, 2019-04-25:
See discussion in https://github.com/arduino-libraries/ArduinoHttpClient/issues/62.

Even though the root cert is flashed for `maker.ifttt.com:443`, `WiFiSSLClient` fails to connect.

@Rocketct do you have time to get a Wireshark capture of this?

cc/ @adriens

adriens, 2019-04-25:
Thanks a lot @sandeepmistry for linking ths issue 👍

Rocketct, 2019-06-20:
@adriens let me inform you that we are in talk with microchip to support this issue, i'll report as soon as possible

adriens, 2019-06-20:
Thank you very much @Rocketct  🙏

Rocketct, 2019-06-27:
hi @adriens with @sandeepmistry have found a fix https://github.com/arduino-libraries/WiFi101/pull/274 could you test and let me know if works on your side ?

sandeepmistry, 2019-07-03:
Fixed via #274.

closed 2019-07-03 by sandeepmistry