arduino-libraries/WiFi101 #271: Is WiFiSSLClient using implicit or explicit SSL?

tuloski, 2019-04-30:
As title. I don't find any documentation about this.

sandeepmistry, 2019-05-01:
Hi @tuloski,

This would not be supported by this library, the same `WiFiClient` underlying socket can't be converted to an SSL connection.

You might have some luck using a WiFiClient with some lower level functions of ArduinoBearSSL: https://github.com/arduino-libraries/ArduinoBearSSL

closed 2019-05-01 by sandeepmistry

tuloski, 2019-05-02:
> This would not be supported by this library, the same `WiFiClient` underlying socket can't be converted to an SSL connection.

How not? There is connectSSL(). Do you mean it cannot be converted on the fly, i.e. connect in clear and then switch to SSL?

sandeepmistry, 2019-05-02:
> Do you mean it cannot be converted on the fly, i.e. connect in clear and then switch to SSL?

Correct, as far as I know the WINC1500 firmware and SDK provided by Microchip/Atmel does not support thing.