arduino-libraries/WiFi101 #273: Cannot connect to ad hoc network

kmontag, 2019-06-13:
I'm hosting an ad hoc network from a Raspberry Pi Zero W with an interface config like the following:

      iface wlan0 inet manual
          wireless-channel 1
          wireless-essid my-network
          wireless-key 1234567890
          wireless-mode ad-hoc
          wireless-ap 02:12:34:56:78:9A

I'm able to connect to this network from my computer, and can use it to e.g. SSH into the Pi.

Using Wifi101, I'm able to see the network when using `scanNetworks()`, but when I try to connect, the connection status never gets past `WL_DISCONNECTED`. Connections to e.g. WPA networks work fine. I see the same behavior regardless of whether there's a key on the ad hoc network.

Here's the relevant sections of the code I'm running, on an Adafruit Feather M0 Wifi (https://www.adafruit.com/product/3010):

```
// Enable this #define if using Adafruit ATWINC (e.g. Feather M0 WiFi).
// If using Arduino Zero + WiFi Shield 101, comment it out:
#define ADAFRUIT_ATWINC

#include <SPI.h>
#include <WiFi101.h>

// Store sensitive data in this file.
#include "secrets.h"

char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                // your network key Index number (needed only for WEP)
int channel = 1;

int led = LED_BUILTIN;
int status = WL_IDLE_STATUS;

WiFiServer server(80);

void printMacAddress();
void listNetworks();

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.println("Access Point Web Server");

  pinMode(led, OUTPUT);      // set the LED pin mode

#ifdef ADAFRUIT_ATWINC
  WiFi.setPins(8, 7, 4, 2); // Pins for Adafruit ATWINC1500 Feather
#endif

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue
    while (true);
  }

  // Print WiFi MAC address:
  printMacAddress();

  // scan for existing networks:
  Serial.println("Scanning available networks...");
  listNetworks();

  // by default the local IP address of will be 192.168.1.1
  // you can override it with the following:
  // WiFi.config(IPAddress(10, 0, 0, 1));

  // print the network name (SSID);
  Serial.print("Connecting to network named: ");
  Serial.println(ssid);

  // attempt to connect to WiFi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Status is: ");
    Serial.print(status);
    Serial.print(". Attempting to connect to network, SSID: ");
    Serial.println(ssid);

    // Have tried with all of these variants....
    //status = WiFi.begin(ssid, keyIndex, pass);
    status = WiFi.begin(ssid, pass);
    //status = WiFi.begin(ssid);

    keyIndex++;
    keyIndex = keyIndex % 4;

    delay(2000);
  }

  // start the web server on port 80
  server.begin();

  // you're connected now, so print out the status
  Serial.print("You're connected to the network");
  printWiFiStatus();
}
```

For the ad hoc network, this just gets stuck at status `6`.

Any ideas, or any other debugging info I can provide?

Rocketct, 2019-06-28:
hi @kmontag i'm tring to reproduce your issue, only setting the network as you described not works!! there is a tutorial that you had follow to set the raspberry? could you share? i have tried also to set the raspberry as AP and from my side works (https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/) if your purpose is only to allow communication through raspberry could be a solution, btw you should also set static ip on the interface to make works the communication without dhcp

kmontag, 2019-06-30:
Thanks @Rocketct. On a fresh Raspbian stretch install, it works for me to create `/etc/network/interfaces.d/wlan0` containing:

```
auto wlan0
iface wlan0 inet manual
      wireless-channel 1
      wireless-essid my-network
      wireless-key 1234567890
      wireless-mode ad-hoc
      wireless-ap 02:12:34:56:78:9A
```

And then add this line to the end of `/etc/dhcpcd.conf` (otherwise DHCPCD tries to auto-configure the interface, which messes it up):

```
denyinterfaces wlan0
```

Then reboot the Pi and you should see the network.

As you mentioned, I can successfully connect to the Pi if it's set up as an AP - but I'm trying to get the Feather connected to a `batman-adv` mesh network shared between a bunch of Pi's, which requires an interface in ad-hoc mode.

Let me know if there's any other info that would be helpful.

Rocketct, 2019-07-22:
hi @kmontag i have make other test, practically it fall in the initial configuration and is not possible set ad-hoc configuration required to allow the communication with the raspberry.
I'm sorry but this feature is not supported by the library.

kmontag, 2019-07-22:
Got it, thanks for looking into it. Just to clarify - do you mean it's not physically possible from a chip like the ATWINC1500? Or is it something that could feasibly be added to the library if someone was willing to work on it?

Rocketct, 2019-07-23:
yes is the library that not support the ad-hoc connection, and there is no planning to support it.

kmontag, 2019-07-29:
Thanks @Rocketct, sorry I think I was unclear. I understand there's no support planned for this - no worries - but I'm curious if this is something that's physically impossible (due to some hardware restriction), or if it could be supported if someone was willing to work on it and submit a PR.

Rocketct, 2019-08-07:
no no don't worry @kmontag checking the datasheet of the module https://datasheet.octopart.com/ATSAMW25H18-MR510PB-Microchip-datasheet-48569841.pdf there is no mention about the ad-hoc network  or p2p infrastructure

kmontag, 2019-08-07:
got it, bummer but thanks for the info!

closed 2019-08-07 by kmontag