arduino-libraries/WiFi101 #275: Server does not respond after lost WiFi connection

der-smu, 2019-07-31:
I am running the WiFiWebServer example with minor modification. My setup:

- Arduino Mega 2560
- Adafruit WINC1500 WiFi shield
- WiFi shield Firmware 19.5.4
- Arduino IDE 1.8.9

I modified the code to check the WiFi status and reconnect to the network, if there's no connection. The code works ok, until the Arduino loses connection. However, after it loses and re-establishes connection to the WiFi network, the server can't be accessed with a browser anymore. Only a power reset seems to solve the issue for me.

    #include <SPI.h>
    #include <WiFi101.h>
    
    
    #include "arduino_secrets.h" 
    ///////please enter your sensitive data in the Secret tab/arduino_secrets.h
    IPAddress ip(xx, xx, xx, xx);
    
    char ssid[] = "mynetwork";    // your network SSID (name)
    char pass[] = "mypassword"; // your network password (use for WPA, or use as key for WEP)
    int keyIndex = 0;                 // your network key Index number (needed only for WEP)
    
    int status = WL_IDLE_STATUS;
    
    
    int count = 0;
    
    WiFiServer server(80);
    
    void setup() {
      //Initialize serial and wait for port to open:
      Serial.begin(9600);
      while (!Serial) {
        ; // wait for serial port to connect. Needed for native USB port only
      }
    
      connectToWiFi();
    
      server.begin();
      // you're connected now, so print out the status:
      printWiFiStatus();
    }
    
    
    void loop() {
      //Serial.println(count);
      count++;
      // listen for incoming clients
      WiFiClient client = server.available();
      if (client) {
        Serial.println("new client");
        // an http request ends with a blank line
        bool currentLineIsBlank = true;
        while (client.connected()) {
          if (client.available()) {
            char c = client.read();
            Serial.write(c);
            // if you've gotten to the end of the line (received a newline
            // character) and the line is blank, the http request has ended,
            // so you can send a reply
            if (c == '\n' && currentLineIsBlank) {
              // send a standard http response header
              client.println("HTTP/1.1 200 OK");
              client.println("Content-Type: text/html");
              client.println("Connection: close");  // the connection will be closed after completion of the response
              client.println("Refresh: 5");  // refresh the page automatically every 5 sec
              client.println();
              client.println("<!DOCTYPE HTML>");
              client.println("<html>");
              // output the value of each analog input pin
              for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
                int sensorReading = analogRead(analogChannel);
                client.print("analog input ");
                client.print(analogChannel);
                client.print(" is ");
                client.print(sensorReading);
                client.println("<br />");
              }
              client.println(count);
              client.println("</html>");
              break;
            }
            if (c == '\n') {
              // you're starting a new line
              currentLineIsBlank = true;
            }
            else if (c != '\r') {
              // you've gotten a character on the current line
              currentLineIsBlank = false;
            }
          }
        }
        // give the web browser time to receive the data
        delay(1);
    
        // close the connection:
        client.stop();
        Serial.println("client disconnected");
      }
    
      if (WiFi.status() != WL_CONNECTED) {
        Serial.println("WiFi not connected. Retry...");
        connectToWiFi();
        printWiFiStatus();
        server.begin();
      }
    
    }
    
    void connectToWiFi() {
      WiFi.config(ip);
      // check for the presence of the shield:
      if (WiFi.status() == WL_NO_SHIELD) {
        Serial.println("WiFi shield not present");
        // don't continue:
        while (true);
      }
      
      // attempt to connect to WiFi network:
      while (WiFi.status() != WL_CONNECTED) {
        Serial.print("Attempting to connect to SSID: ");
        Serial.println(ssid);
        // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
        status = WiFi.begin(ssid, pass);
    
        // wait 10 seconds for connection:
        delay(10000);
      }
      
    }
    
    void printWiFiStatus() {
      // print the SSID of the network you're attached to:
      Serial.print("SSID: ");
      Serial.println(WiFi.SSID());
    
      // print your WiFi shield's IP address:
      IPAddress ip = WiFi.localIP();
      Serial.print("IP Address: ");
      Serial.println(ip);
    
      // print the received signal strength:
      long rssi = WiFi.RSSI();
      Serial.print("signal strength (RSSI):");
      Serial.print(rssi);
      Serial.println(" dBm");
    }

der-smu, 2019-08-05:
FYI, I recognized that I need to turn off the WiFi shield with `WiFi.end()` before reconnecting to it. Using `client.stop()` just after `status = WiFi.begin(ssid, pass)` also worked for this example, but not for another scenario.

I do not know whether this is behaviour is by design, anyhow I didn't know about it.

Rocketct, 2019-08-06:
hi @der-smu i'm running your example from yesterday and forcing the disconnection it properly reconnect to the AP it expose correctly his ip on the network.
For my test, i'm using the module version 19.6.1, and 19,5,4, and the library Wifi101 0.14.0 and 0.16.0.

The WiFi.end() should be called just after the client stop, as follow:

```c++
 if (WiFi.status() != WL_CONNECTED) {
    Serial.println("WiFi not connected. Retry...");
    WiFi.end();
    connectToWiFi();
    printWiFiStatus();
    server.begin();
  }
```

der-smu, 2019-08-07:
Hi @Rocketct, are you able to access the sensor readings from the browser **after** it reconnects? Here, the Arduino reconnects correctly to the network and I am also able to ping it from my computer, but the server part is not working correctly.
My WiFi Shield is on version 19.5.4. and the WiFi101 library on version 0.16.0.

Just to be clear, the web server is working correctly, if I use WiFi.end() in the way you posted it.

Rocketct, 2019-08-07:
Yes i'm able to read the value on the pin A0-A6, from the browser, like:

```
analog input 0 is 380
analog input 1 is 420
analog input 2 is 417
analog input 3 is 211
analog input 4 is 305
analog input 5 is 271
6021587 
```
in my experiment i'm force the re-connection, making a power cycle of the modem, i'm using an arduino mkr1000 to make this test, you should try to update the f/w module to do it you have to ask to the seller

sandeepmistry, 2019-08-07:
Hi @Rocketct, @der-smu,

`WiFi.end()` powers off the module. It should not be needed for the server to work if `server.begin()` is called after the reconnection. @Rocketct we should investigate the root cause further.

der-smu, 2019-08-23:
Maybe it is related to this issue: I found that the web server also does not respond, after the Arduino lost power. I always have to manually press the Reset-Button, in order to reach the Arduino from a browser. However, I can't yet reproduce this issue with a minimal working example.

Rocketct, 2019-08-27:
Hi @der-smu, in my test i don't use the `WiFi.end()`, in fact the server should work without the .end() API, have you any news about this issue?

der-smu, 2019-08-27:
Hi @Rocketct, I just failed to reproduce this issue. The code in my original post works just fine. However, I have no clue what could've cause the troubles. I can just assume that the issue was not on the Arduino, but on my computer/browser.

Rocketct, 2019-08-27:
ok @der-smu then i'll close the issue if you need, feel free, in future to reopen the issue.

closed 2019-08-27 by Rocketct