arduino-libraries/WiFi101 #277: Can't connect to a server using TLS_ECDHE_RSA_WITH_AES

pictorf, 2019-08-18:
Hi,

Here is the problem already exposed in the arduino forum.
https://forum.arduino.cc/index.php?topic=628993.new#new

I try to connect a MKR1000 to tgftp.nws.noaa.gov.
It appears this server uses TLS_ECDHE_RSA_WITH_AES.
The MKR100 firmware was updated and the domain name was added to the list of certificates.
Despite I turned on the EC cipher in WiFi101 library, I can't establish a connection.

sandeepmistry, 2019-08-26:
Hi @pictorf,

The `TLS_ECDHE_RSA_WITH_AES` cipher will not work with this library, so this is why it's not enabled by default. When it is enabled, it requires integration with an ECC508a or ECC608a chip, something we don't plan to support with this library.

Would you be able to provide a list of ciphers that server supports?

pictorf, 2019-08-28:
Hi @sandeepmistry ,

In the forum topic I opened and did indicate above, I was told the server I try to connect to (https://tgftp.nws.noaa.gov) uses TLS_ECDHE_RSA_WITH_AES.

A quick try modifying line 317 as indicated in your post above doesn't work.

After opening an issue here on GitHub, I looked at other issues and found some people use ArduinoBearSSL.
I gave it a try and my sketch works now.
The draw back is that my sketch is more than 3 times bigger in memory.

sandeepmistry, 2019-09-18:
For further assistance with this I suggest you contact Microchip/Atmel support for this.

Please re-open with the ticket number if you decide to move forward and contact them.

closed 2019-09-18 by sandeepmistry