arduino-libraries/WiFi101 #278: Bug in WiFiCocket.cpp causing infinite loop

S2Doc, 2019-09-03:


sandeepmistry, 2019-09-18:
Hi @S2Doc,

Do you have steps to reproduce this condition? What f/w is your WINC1500 running?

Another option to your proposal would be to have a retry timeout.

S2Doc, 2019-09-18:
I have experienced this problem when sending MQTT packets at 3-5 second intervals.  It occurs after as few as 2 packets have been sent or only after tens of thousands.  It does not seem to correlate with losing the MQTT connection, although it is impossible to be totally certain of this.

The problem occurs with WINC1500 firmware v19.4.4 and v19.5.4.  I have not tested with other versions.

I have had good results by counting the number of times through the while loop and exiting after the 10th try.  Using a timeout to exit the loop would be a good alternative as long as the length of the timeout was adjustable, perhaps through a #define, so that any watchdog timer set by the user will not cause a reset before the timeout is reached.

S2Doc, 2019-09-18:
Note that there are other places in WiFiSocket.cpp with loops that may never exit.   Another is in WiFiSocket::Connect(), see [#279](https://github.com/arduino-libraries/WiFi101/issues/279).

WiFiSocketClass::bind and WiFiSocketClass::listen also have hard-coded timeouts which may cause problems if the user sets a watchdog timer to reset before the timeout has occurred.