arduino-libraries/WiFi101 #281: Possible to add custom WiFi101/WiFiNINA updater

GregTerrell, 2019-11-05:
If I have been working on an updated WINC1500 firmware image for the purpose of creating a custom provisioning web page (similar to what Arduino has already done) is it possible to add it to the to firmware image list in the IDE's firmware updater?

Is the firmware updater source available anywhere to build an outboard updater that leverages the FirmwareUpdater.ino?

Thanks
Greg Terrell
LooUQ

sandeepmistry, 2019-11-11:
Hi @GregTerrell,

We won't likely accept a new firmware image from you from you in the official distribution.

You can build a custom version of the updater and manually place in the correct location in the sketch book for your purpose.

The source code is here: https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin

closed 2019-11-11 by sandeepmistry

GregTerrell, 2019-11-11:
Thank you for the guidance, this information is exactly what I was looking for. The firmware I am working on is not for general consumption, i.e. specific to a customer. But reusing the firmware updater source will save consider time. 
Thanks again!
Greg