arduino-libraries/WiFi101 #282: Https Web Server configuration works, but client.remoteIP() returns IP 0.0.0.0 and client.remotePort() returns 0

norfig, 2019-11-07:
Verified that the example HTTP Web Server code (in File->Examples->WiFi101->WWWWebServer) works fine. I added calls to client.remoteIP() and client.remotePort() to display the IP address and port number of the remote host and it works fine.

Modified the code to run as an HTTPS Web Server (with the two simple changes listed below) and it also works fine. However, client.remoteIP() always returns IP 0.0.0.0 and client.remotePort() always returns 0.

Verified that WiFi WiFiSocketClass::handleEvent() receives SOCKET_MSG_ACCEPT with IP and port equal to 0 when HTTPS is enabled. Is there another way to get the remote IP and port?

Code changes to make if an HTTPS Web server (as .
WiFiServer server(80);
changed to:
WiFiServer server(443);

server.begin();
changed to:
server.beginSSL();

Hardware:
ARDUINO MEGA 2560 REV3
Adafruit WINC1500 WiFi Shield

sandeepmistry, 2019-11-11:
Hi @norfig,

Unfortunately HTTPS server is not something that is fully supported at this time. I would suggest to contact Microchip support for further assistance.

norfig, 2019-12-01:
Microship support was able to reproduce the issue and is looking into it.

norfig, 2019-12-30:
Microship support indicated that it created a patch that fixes the issue and provide it to the Microship development team. According to Microship support, the fix should be part of an upcoming firmware release. However, they currently did not know when such new firmware would be released.

closed 2019-12-30 by norfig