arduino-libraries/WiFi101 #283: Firmware for the Wifi 101 Shield?

TheShadowhawke, 2019-12-08:
So, I just purchased the Adafruit version of the Wifi 101 shield. It apparently is a WINC1501 Model A. When running the firmware checks, it says that the latest version of the firmware is 19.6.1, but I only see 19.4.4 available. Am I misunderstanding something in how to do the upgrade? Or is there not the newer version for this board? Thanks!

Rocketct, 2020-02-26:
Yes the model A is supported only up to the 19.4.4 version! can i ask  which is exactly the model of adafruit shield you bought?

TheShadowhawke, 2020-02-26:
I apologize, but I have long since returned the shield to the Amazon vendor, and have since gotten a working one from Adafruit directly. Thank you for looking into it though.

closed 2020-02-26 by TheShadowhawke