arduino-libraries/WiFi101 #284: getTime appears to be returning the wrong date

tigoe, 2020-01-11:
WIFi.getTime() appears to be returning the wrong date by one day. 

Using [this code](https://github.com/ITPNYU/clock-club/blob/master/Microcontroller_Time_Setting_Methods/WiFiTimeSet/WiFiTimeSet.ino), I should be getting 11/1/2020 today, but I am getting 10/1/2020. I'm not clear on why, but the time is correct.

Running this on a MKR1000, library version 0.16.0

S2Doc, 2020-01-14:
I'm having the same or similar problem.  3 or 4 out of 100 devices I have in the field are suddenly showing times that are exactly 24 hours before the actual time.  

Using Adafruit Feather M0 with WINC 15000.

This is causing  significant problems.  Any help would be greatly appreciated.

S2Doc, 2020-02-05:
In my case the problem turned out to be that the ATWINC1500 wifi module was running firmware v19.4.4.   Microchip's revision notes indicate that version had problems of some sort with NTP servers.  Upgrading the module's firmware to v19.5.4 or higher from the Arduino IDE fixed the problem.  

Let us know if that works for you.

tigoe, 2020-02-07:
The problem persists on the ESP32 module on the MKR1010 and Nano33 IoT boards, though

S2Doc, 2020-02-07:
The ESP32 doesn't use WiFi101, does it?

tigoe, 2020-02-14:
No, but the WiFi101 and the WiFiNINA modules share the same WiFi core API. Posted it here, but had hit the problem on both.

Rocketct, 2020-02-26:
Hi @tigoe i tried the sketch in the the wifi1010 firmware v1.3.0 and i get the same results of the fixed mkr1000 tomorrow when i'll get i'll try also the 33 iot

Rocketct, 2020-02-28:
@tigoe run on  33 Iot the and works fine too

yanliu888, 2021-06-27:
Same issue as @tigoe mentioned here - it suddenly changed to a wrong time during measuring. How to fix it? I don't see a higher version of firmware in Arduino software for NANO 33 IoT board - the latest version is version 1.4.5. Thank you very much in advance!

yanliu888, 2021-06-27:
I just figured out maybe the example code you are using sets:

`set_hour = rtc_wifi.getHours() + GMT;`

but set_hour cannot be negative, so can fix the code like this by deleting GMT in the above line and converting UTC to GMT like this:

`rtc_wifi.setEpoch(epoch + GMT * 60 *60);`

tigoe, 2021-06-28:
Thanks, I'll give that a go. I had found a way around this awhile ago, but
it's always nice to see a different solution, so thank you for that.

per1234, 2025-03-20:
Closing as it appears to have been caused by a bug in the ATWINC1500 firmware, not in the WiFi101 library codebase.

closed 2025-03-20 by per1234