arduino-libraries/WiFi101 #294: What is the purpose of 'Device name' field in the Provisioning Page

broddo, 2020-07-25:
This isn't really an issue but I thought I'd ask here nonetheless. 

I noticed that Atmel's default provisioning page has a 'Device name' field. The field is indeed used by the html page (i.e. its not just a dummy field) and gets sent as part of the POST request to the WINC1500. 

However, there's no mention of this field in the struct tstrM2MProvisionInfo and examining the contents pvMsg in my debugger doesn't show any other unhandled fields. Does anyone know what the intension of this field is and if it can be read from the M2M library? 

The reason I ask is that I was hoping to hijack it in order to allow a user send more settings to my device than just the SSID and enc-key. 

Thanks!

avispeng, 2021-07-26:
If you go to the dashboard of your wifi router, you can find a list of connected devices. The device name is shown here. It helps you to identify the device.

broddo, 2021-07-26:
I'll modify the title as it doesn't really match what I'm asking about. I'd like to know how the 'device name' field is handled in the M2M driver. I don't see it being referenced anywhere. The driver does provide an API for updating the Device Name from the firmware directly but it doesn't appear to make use of the 'device name' field at all.