arduino-libraries/WiFi101 #301: bitwise and operator used on sequential enum values

denisbaylor, 2020-12-19:
Throughout WiFi.cpp, the _status variable is tested using the bitwise and operator, but its possible values are not bitmasks.  This looks wrong.

For example, at line 450 we have
`if (!(_status & WL_CONNECTED)) {`

The value of WL_CONNECTED is 3, so the expression (_status & WL_CONNECTED) is true if either of the lowest order 2 bits is set in _status.  That makes line 450 behave the same as:

`if (!(_status == WL_NO_SSID_AVAIL || _status == WL_SCAN_COMPLETED || _status == WL_CONNECTED || _status == WL_CONNECTION_LOST || _status == WL_DISCONNECTED || _status == WL_AP_LISTENING || _status == WL_AP_FAILED || _status == WL_PROVISIONING || _status == WL_PROVISIONING_FAILED)) {`

And since line 446 looks like it intended to break out of the loop when _status was either WL_CONNECTED or WL_DISCONNECTED (but not quite because the same bug exists there), then line 450's if statement never succeeds because it doesn't distinguish between those two values.  I believe the intended behavior of line 450 is instead

`if (!(status_ == WL_CONNECTED)) {`

or the simpler

`if (status != WL_CONNECTED) {`

jandrassy, 2022-01-09:
it is strange, but it doesn't create a bug since only one enum value is assigned to status_ at time

denisbaylor, 2022-01-12:
It doesn't matter that only one enum value is assigned to status_ at a time.

If I do:
   status_ = WL_CONNECTED;

And then later do
   if (status_ & WL_DISCONNECTED)

That condition will be true.  This is because WL_CONNECTED is 3, WL_DISCONNECTED is 6,
and both of those values in binary have the 2 bit on.

jandrassy, 2022-01-12:
you are right