arduino-libraries/WiFi101 #305: Wifi101 doesn't work once add the freeRTOS library

samsonswi, 2021-01-13:
Hi  ,

I'm working on project with STM32429Zi Nucleo board to connect to wifi, for that I use ATWINC1500 and use wifi101 library. I use platformio with Arduino framework.

First what I did was , create a new project .Add the wifi101 library and use the example ping sketch .
it works (pinging the given domain).
Then I add the STM32duino freeRTOS library to the same project without even modifying the existing sketch, and it doesn't work .
Stuck at " Attempting to connect to WPA SSID: ABC "
I talk this issue with the platformio community and they said the issue is with the wifi101 library and need to issue a complain with them .
Appreciate if someone can help.

Thanks,
Samson.

per1234, 2021-01-13:
>I talk this issue with the platformio community and they said the issue is with the wifi101 library

Please share the link to that discussion in case it will provide some useful context.

samsonswi, 2021-01-13:
Hi ,
please check the bottom of the thread ,

https://community.platformio.org/t/stm32duino-freertos-doesnt-work/18505/7

thanks,
Samson.

maxgerhardt, 2021-02-02:
Maybe related to a dependent library? 

https://github.com/stm32duino/STM32Ethernet/issues/36