arduino-libraries/WiFi101 #328: Multiple compilation problems on ESP8266

max5555, 2022-01-29:
I cannot compile any of my old projects. I tried to compile on of the examples and got the same errors

`Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (3.2.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)    
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 111 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <WiFi101> 0.16.1
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio\build\d1_mini\src\main.cpp.o
Generating LD script .pio\build\d1_mini\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\d1_mini\lib6f8\SPI\SPI.cpp.o
Compiling .pio\build\d1_mini\libfa1\WiFi101\WiFi.cpp.o
Compiling .pio\build\d1_mini\libfa1\WiFi101\WiFiClient.cpp.o
In file included from C:\users\y\.platformio\lib\WiFi101\src/common/include/nm_common.h:45,
                 from C:\users\y\.platformio\lib\WiFi101\src/socket/include/socket.h:60,
                 from C:\users\y\.platformio\lib\WiFi101\src\utility/WiFiSocket.h:24,
                 from C:\users\y\.platformio\lib\WiFi101\src\WiFi.cpp:36:
C:\users\y\.platformio\lib\WiFi101\src/bsp/include/nm_bsp.h:220:6: error: variable or field 'nm_bsp_sleep' declared void
  220 | void nm_bsp_sleep(uint32 u32TimeMsec);
      |      ^~~~~~~~~~~~
C:\users\y\.platformio\lib\WiFi101\src/bsp/include/nm_bsp.h:220:19: error: 'uint32' was not declared in this scope; did you mean 'uint16'?
  220 | void nm_bsp_sleep(uint32 u32TimeMsec);
....`

main.cpp
`#include <Arduino.h>
/*

 This example  prints the WiFi 101 Shield's MAC address, and
 scans for available WiFi networks using the WiFi 101 Shield.
 Every ten seconds, it scans again. It doesn't actually
 connect to any network, so no encryption scheme is specified.

 Circuit:
 * WiFi 101 Shield attached

 created 13 July 2010
 by dlf (Metodo2 srl)
 modified 21 Junn 2012
 by Tom Igoe and Jaymes Dec
 */

#include <SPI.h>
#include <WiFi101.h>

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi 101 Shield not present");
    // don't continue:
    while (true);
  }

  // Print WiFi MAC address:
  printMacAddress();

  // scan for existing networks:
  Serial.println("Scanning available networks...");
  listNetworks();
}

void loop() {
  delay(10000);
  // scan for existing networks:
  Serial.println("Scanning available networks...");
  listNetworks();
}

void printMacAddress() {
  // the MAC address of your WiFi 101 Shield
  byte mac[6];

  // print your MAC address:
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  printMacAddress(mac);
}

void listNetworks() {
  // scan for nearby networks:
  Serial.println("** Scan Networks **");
  int numSsid = WiFi.scanNetworks();
  if (numSsid == -1)
  {
    Serial.println("Couldn't get a WiFi connection");
    while (true);
  }

  // print the list of networks seen:
  Serial.print("number of available networks:");
  Serial.println(numSsid);

  // print the network number and name for each network found:
  for (int thisNet = 0; thisNet < numSsid; thisNet++) {
    Serial.print(thisNet);
    Serial.print(") ");
    Serial.print(WiFi.SSID(thisNet));
    Serial.print("\tSignal: ");
    Serial.print(WiFi.RSSI(thisNet));
    Serial.print(" dBm");
    Serial.print("\tEncryption: ");
    printEncryptionType(WiFi.encryptionType(thisNet));
    Serial.flush();
  }
}

void printEncryptionType(int thisType) {
  // read the encryption type and print out the name:
  switch (thisType) {
    case ENC_TYPE_WEP:
      Serial.println("WEP");
      break;
    case ENC_TYPE_TKIP:
      Serial.println("WPA");
      break;
    case ENC_TYPE_CCMP:
      Serial.println("WPA2");
      break;
    case ENC_TYPE_NONE:
      Serial.println("None");
      break;
    case ENC_TYPE_AUTO:
      Serial.println("Auto");
      break;
  }
}

void printMacAddress(byte mac[]) {
  for (int i = 5; i >= 0; i--) {
    if (mac[i] < 16) {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0) {
      Serial.print(":");
    }
  }
  Serial.println();
}
`

platformio.ini
`[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
`

max5555, 2022-01-29:
The same problem in Arduino IDE

per1234, 2022-01-29:
HI @max5555. As it says [in the readme](https://github.com/arduino-libraries/WiFi101#readme):

>WiFi101 library for for the Arduino WiFi Shield 101 and MKR1000 board

This library is not written for use with the ESP8266. You will want to use the ESP8266WiFi library for that board:
https://arduino-esp8266.readthedocs.io/en/3.0.2/libraries.html#wifi-esp8266wifi-library

If you would like further assistance, feel free to ask on the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2022-01-29 by per1234

max5555, 2022-01-29:
I don't use it directly. It is compiled as a dependency
[image]

max5555, 2022-01-29:
And it was working before without any problem