arduino-libraries/WiFi101 #330: Library does not work with WINC1500 and Mega2560

rkern62, 2022-02-28:
Unfortunately the library seems not to work correctly using the WINC1500 shield with a Mega2560 board.
While the shield is reported as available, the firmware at 19.5.4 it is not possible to list the available networks (reports 0) as well as no connection to a specific WPA Access Point is possible.
If put into AP mode, it shows up on the list of available networks.

In that state I can not use the Wifi shield at all :-(.

Any help will be greatly appreciated.
Greetings, Ruediger Kern.

jandrassy, 2022-03-01:
for me it works

rkern62, 2022-03-01:
Many thanks for your fast answer! :-) I tried with the version you pointed to, with the same results (no networks being found).
Is there any possibility for debugging the library in more depth? Are there any other obstacles that I might have been missing?

Many thanks for your support and your valuable time,
greetings from germany, yours Ruediger Kern.

rkern62, 2022-03-01:
Yes, I got the code from the https://github.com/jandrassy/WiFi101/tree/all_fixes link.
As I changed things in a hurry, I'll double check again and give you feedback.
Beside this change, is there any option for enabling debugging for getting some more details of what is going on?
THX a lot! :-)

jandrassy, 2022-03-01:
you can add Serial printing in any cpp file if you include Arduino.h

rkern62, 2022-03-01:
I tried this already but I found I probably need debugging a level deeper. So I found the M2M_ERR statements there but had no success in redefining the printf in nm_debug.h.

jandrassy, 2022-03-03:
sorry, it is not my library. I just try to help with it a little

jandrassy, 2022-03-03:
@per1234 Per, it is solved

closed 2022-03-03 by per1234