arduino-libraries/WiFi101 #354: ATWINC1500 with WiFi101 with MQTT always leads to a dead connection after some time

kongomongo, 2024-01-23:
Hi everyone,

I am writing a tool on an Arduino Mega (Controllino MAXI is the exact hardware because it has builtin relays) that simply allows me to open and close windows using relays.

Additionally the Controllino should send and receive MQTT topics/commands to monitor and modify the state of the windows.

Now the problem with the Controllino is that it does not have wireless capability. So I started looking at different solutions to this problem and had to discard quite a number of them because of low connection speeds.

I found the Adafruit ATWINC1500 and it seemed like a good fit for my case, enabling wireless on the controllino.

The project had worked on wire flawlessly. After converting it to wireless the connection kept dropping out after a couple of days. So I added quite a number of debug logging via serial. 

I kept adding error handling and recovery into the programming but now I am at a loss what the root cause of the problems is.

So the latest code shows this logging info:

```
...
04:29:00.527 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:29:05.554 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:29:05.554 -> MQTT publish {"DF_K_Auto":0,"DF_K_Auto_Timer":0,"DF_E_Auto":0,"DF_E_Auto_Timer":0,"RSSI":-39}
04:29:10.550 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:29:15.523 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:29:15.566 -> MQTT publish {"DF_K_Auto":0,"DF_K_Auto_Timer":0,"DF_E_Auto":0,"DF_E_Auto_Timer":0,"RSSI":-39}
04:30:28.030 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:30:28.080 -> MQTT connection has gone offline!
04:30:28.080 -> mqtt trying connection
04:30:48.054 -> Connection to MQTT failed, returnCode: 0 LWMQTT_CONNECTION_ACCEPTED // lastError: -3 LWMQTT_NETWORK_FAILED_CONNECT
04:30:48.097 -> MQTT Reconnect Interval set to: 60000
04:30:48.097 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:30:53.069 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
<snip>
04:31:38.101 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:31:43.080 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:31:48.053 -> No MQTT connection and Reconnect Interval has expired
04:31:48.098 -> mqtt trying connection
04:32:08.074 -> Connection to MQTT failed, returnCode: 0 LWMQTT_CONNECTION_ACCEPTED // lastError: -3 LWMQTT_NETWORK_FAILED_CONNECT
04:32:08.074 -> MQTT Reconnect Interval set to: 120000
04:32:08.074 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:32:13.100 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
<snip>
04:33:58.130 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:34:03.093 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:34:08.085 -> No MQTT connection and Reconnect Interval has expired
04:34:08.085 -> mqtt trying connection
04:34:28.102 -> Connection to MQTT failed, returnCode: 0 LWMQTT_CONNECTION_ACCEPTED // lastError: -3 LWMQTT_NETWORK_FAILED_CONNECT
04:34:28.102 -> MQTT Reconnect Interval set to: 180000
04:34:28.102 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:34:33.079 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
<snip>
04:37:18.118 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:37:23.112 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:37:28.098 -> No MQTT connection and Reconnect Interval has expired
04:37:28.098 -> mqtt trying connection
04:37:48.113 -> Connection to MQTT failed, returnCode: 0 LWMQTT_CONNECTION_ACCEPTED // lastError: -3 LWMQTT_NETWORK_FAILED_CONNECT
04:37:48.113 -> MQTT Reconnect Interval set to: 240000
04:37:48.113 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:37:53.105 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
<snip>
04:41:38.132 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:41:43.158 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:41:48.092 -> No MQTT connection and Reconnect Interval has expired
04:41:48.092 -> mqtt trying connection
04:42:08.114 -> Connection to MQTT failed, returnCode: 0 LWMQTT_CONNECTION_ACCEPTED // lastError: -3 LWMQTT_NETWORK_FAILED_CONNECT
04:42:08.114 -> MQTT Reconnect Interval set to: 300000
04:42:08.114 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:42:13.117 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
<snip>
04:46:58.145 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
04:47:03.165 -> SSID: My_Wifi_SSID, IP Address: 192.168.1.104, Gateway IP Address: 192.168.1.1, WiFi status: 3 WL_CONNECTED, signal strength (RSSI):-39 dBm
[... 874 lines cut ...]
      return i;
  }

  return -1;
}

int findMqttAutoTimerByName(String &name)
{
  return findMqttAutoTimerByName(name.c_str());
}

int findMqttAutoTimerByName(const char* name)
{
  for (int i=0; i<NUM_WINDOWS; i++)
  {
    if (strcmp(mqttNameAutoTimer[i], name) == 0)
      return i;
  }

  return -1;
}

```

Crossposting this from https://forum.arduino.cc/t/atwinc1500-with-wifi101-with-mqtt-always-leads-to-a-dead-connection-after-some-time/1213245

per1234, 2024-01-23:
Hi @kongomongo. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there.

closed 2024-01-23 by per1234

kongomongo, 2024-01-24:
@per1234 thanks for your feedback. So .disconnect() and subsequent .begin() / .connect() should not fail as in my example? (I wasn't sure if this might be a library bug, people have opened issues like https://github.com/arduino-libraries/WiFi101/issues/251 which seem similar).

Is there any form of debugging that can be enabled to troubleshoot?

Many thanks in advance!